diablodale/jit.openni #33: Crash when grabbing the usermap for a long time

aalex, 2015-02-25:
It seems like Max crashes after a few minutes if we grab the usermap.

```
Process:               Max [471]
Path:                  /Applications/Max 6.1/Max.app/Contents/MacOS/Max
Identifier:            com.cycling74.Max
Version:               6.1.9 [c65762b] (6.1.9)
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           Max [471]
User ID:               501

Date/Time:             2015-02-25 12:06:06.497 -0500
OS Version:            Mac OS X 10.10.2 (14C109)
Report Version:        11
Anonymous UUID:        216BE4EB-3DB0-0BFE-30D7-932BCBDD2A3E

Sleep/Wake UUID:       59DB3886-B354-4F1F-AB32-8697E8A1BF82

Time Awake Since Boot: 3100 seconds
Time Since Wake:       33 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x000000000c03ca8e

VM Regions Near 0xc03ca8e:
    MALLOC_TINY            000000000b900000-000000000c000000 [ 7168K] rw-/rwx SM=PRV  
--> mapped file            000000000c000000-000000000c0bd000 [  756K] r--/rwx SM=COW  /Library/Fonts/Arial.ttf
    mapped file            000000000c0bd000-000000000c5e2000 [ 5268K] r--/rwx SM=COW  /System/Library/Fonts/HelveticaNeueDeskInterface.ttc

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_malloc.dylib        	0x93a80050 szone_free + 3556
1   com.apple.CoreFoundation      	0x9602f7f8 __CFAllocatorSystemDeallocate + 24
2   com.apple.CoreFoundation      	0x96018a34 CFAllocatorDeallocate + 100
3   com.apple.CoreFoundation      	0x9602cf86 CFRelease + 1862
4   com.apple.CoreFoundation      	0x960939fd CFRunLoopAddTimer + 477
5   com.apple.Foundation          	0x95c99ae7 -[NSObject(NSDelayedPerforming) performSelector:withObject:afterDelay:inModes:] + 991
6   com.apple.AppKit              	0x92b7c020 +[_NSAutomaticFocusRing setNeedsUpdateForView:] + 284
7   com.apple.AppKit              	0x92b7bcf4 -[NSView noteFocusRingMaskChanged] + 63
8   com.apple.AppKit              	0x933322d3 NSViewSetNeedsDisplayInRect + 1159
9   com.apple.AppKit              	0x92b7adf0 -[NSView setNeedsDisplayInRect:] + 35
10  com.cycling74.Max             	0x0046a0a2 juce::NSViewComponentPeer::repaint(juce::Rectangle<int> const&) + 246
11  com.cycling74.Max             	0x003839c5 juce::Component::internalRepaint(int, int, int, int) + 337
12  com.cycling74.Max             	0x00383926 juce::Component::internalRepaint(int, int, int, int) + 178
13  com.cycling74.Max             	0x00383926 juce::Component::internalRepaint(int, int, int, int) + 178
14  com.cycling74.Max             	0x00383926 juce::Component::internalRepaint(int, int, int, int) + 178
15  com.cycling74.Max             	0x00383926 juce::Component::internalRepaint(int, int, int, int) + 178
16  com.cycling74.Max             	0x00383926 juce::Component::internalRepaint(int, int, int, int) + 178
17  com.cycling74.Max             	0x00383926 juce::Component::internalRepaint(int, int, int, int) + 178
18  com.cycling74.Max             	0x00383926 juce::Component::internalRepaint(int, int, int, int) + 178
19  com.cycling74.Max             	0x00383926 juce::Component::internalRepaint(int, int, int, int) + 178
20  com.cycling74.Max             	0x00383926 juce::Component::internalRepaint(int, int, int, int) + 178
21  com.cycling74.Max             	0x001191ca BoxComponent::repaintCanvasInset(double) + 102
22  com.cycling74.Max             	0x001f0667 PatcherComponent::boxnotify_redraw(object*) + 75
23  com.cycling74.Max             	0x002225c9 patcherview_boxnotify(_patcherview*, symbol*, short, atom*) + 382
24  com.cycling74.Max             	0x000c8809 object_method_typedfun + 132
25  com.cycling74.Max             	0x000c897c object_method_typed + 177
26  com.cycling74.Max             	0x0019b256 jpatcher_call_object_method_typed + 48
27  com.cycling74.Max             	0x0006138a linklist_funall_imp + 102
28  com.cycling74.Max             	0x00061434 linklist_funall + 39
29  com.cycling74.Max             	0x001a9e77 jpatcher_doboxnotify + 444
30  com.cycling74.Max             	0x00038e29 defer + 54
31  com.cycling74.Max             	0x001a9cb3 jpatcher_boxnotify + 88
32  com.cycling74.Max             	0x0019b0b3 jbox_boxnotifyredrawifspooled + 70
33  com.cycling74.Max             	0x0006138a linklist_funall_imp + 102
34  com.cycling74.Max             	0x00061434 linklist_funall + 39
35  com.cycling74.Max             	0x0019b0ef jpatcher_redrawqfn + 44
36  com.cycling74.Max             	0x0010827e sysqelem_dequeue + 134
[... 685 lines cut ...]
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x135), Broadcom BCM43xx 1.0 (7.15.159.13.12)
Bluetooth: Version 4.3.2f6 15235, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en2
PCI Card: AMD FirePro D700, Display Controller, Slot-1
PCI Card: AMD FirePro D700, Display Controller, Slot-2
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: USB3.0 Hub
USB Device: Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Hub
USB Device: USB Keyboard
USB Device: USB2.0 Hub
USB Device: Hub
USB Device: Hub
USB Device: Hub
USB Device: USB Optical Mouse
USB Device: Hub
USB Device: Xbox NUI Camera
USB Device: Xbox Kinect Audio, © 2011 Microsoft Corporation. All rights reserved.
USB Device: Xbox NUI Motor
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
```

diablodale, 2015-02-25:
Apple bought and then killed NITE, the part of OpenNI that did body detection. It is illegal for anyone anywhere at anytime to distribute, install, or use NITE. As far as I know, NITE is needed to run jit.openni which suggests illegal usage.

I do am not providing any support for jit.openni due to Apple killing the NITE project. It is a dead-end technology. I recommend you transition to modern supported technologies like Kinect for Windows and dp.kinect.

aalex, 2015-02-25:
Ok. Thank you for the info. Sorry for posting bug reports for this dead project.

closed 2015-02-25 by aalex

diablodale, 2015-02-25:
No worries.  FYI, in teh thread that crashed (thread 0), I don't even see calls to jit.openni, Openni, or NITE. It likely something else that is crashing.